jotego/jtcps #154: Provide a "Demo Sounds: Off" option

elrubioVS, 2023-01-02:
Is it possible to add a feature to turn off demo sounds for all CPS games?

Would love to have the CPS games cycling on demo but can't so this with the sound on.

jotego, 2023-02-07:
You have to do it manually at the moment. You can change the MRA files or the configuration files that MiSTer reads to achieve this, but it isn't a trivial thing to do if you're not familiar with how these files work.

There is no means of changing the DIP switches of multiple games in MiSTer (nor MAME) and it isn't trivial how to do it because there are few situations where you can do it.

Having demo sounds off by default usually results in people thinking the core is broken because there is no sound. Thus, we have to set it on by default.

I think your best option is just to mute your TV. Sorry about that.

closed 2023-02-07 by jotego

elrubioVS, 2023-02-09:
Thanks for the response Jotego! Do you have the specific dip switch setting for a typical CPS MRA file?

I would like to test this with a copy of the Aliens VS Predator MRA e.g.

```
<switches default="FF,FF,C9">
  <dip bits="15"    name="Cabinet" ids="Cocktail,Upright"/>`
  <dip bits="16,17" name="Coinage" ids="2c/1cr,1c/1cr,1c/2cr,Free Play" values="3,1,2,0"/>
  <dip bits="18,19" name="Lives" ids="1,2,3,5"/>
  <dip bits="20,21" name="Bonus Life After" ids="10000,15000,20000,None"/>
  <dip bits="22"    name="Difficulty" ids="Hard,Normal"/>
</switches>
```

jotego, 2023-02-10:
Just open the MRA files and look at what's in there. For CPS2 games, there are no DIP switches and you have to go into the game service menu, set the option and save it. That will create a NVRAM file with that information. The CPS2 MRA files contain default settings for the NVRAM file in a long hexadecimal string. You can look into changing that string with the one you obtained from dumping the NVRAMs.
As I said, there is no quick way of doing this.